Turge08/print_area_bed_mesh #4: TypeError: unsupported operand type(s) for -: 'str' and 'int'

WickedInvi, 2021-11-25:
Followed the instructions. Getting the following error.

```
14:55:56 
File opened:toolbox-4_1h32m_0.2mm_235C_PLA.gcode Size:722883
14:55:56 
File selected
14:56:41 
// print_min: 70.9073,54.9073
14:56:41 
// print_max: 159.093,159.093
14:56:41 
// mesh_min: 50.9073,34.9073
14:56:41 
// mesh_max: 179.093,179.093
14:56:41 
// probe_count: 3, 7
14:56:41 
!! Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': TypeError: unsupported operand type(s) for -: 'str' and 'int'
14:56:41 
!! Error evaluating 'gcode_macro BED_MESH_CALIBRATE:gcode': TypeError: unsupported operand type(s) for -: 'str' and 'int'
```

Also removed variable_parameter_PRINT_MIN : 0,0 and variable_parameter_PRINT_MAX : 0,0 as they exist in the config are they needed in the START_PRINT macro?
Result with and without are as above.

Turge08, 2021-11-25:
Yes, they're needed in both macros since the slicer passes the arguments to the start_print macro and the start_print macro passes the argument to the bed_mesh_calibrate macro.

I just (hopefully) fixed the error. Can you please update the macro and try again?

If you're still having issues, can you attach your klippy.log?

WickedInvi, 2021-11-25:
!! bed_mesh: invalid probe_count option when using bicubic interpolation. Combination of 3 points on one axis with more than 6 on another is not permitted. Configured Probe Count: 3, 7

Using a 7x7 as default. You will have to add checks for it.

Turge08, 2021-11-27:
@WickedInvi  I'm looking into this. Someone else just reported the issue and I'll be changing the minimum probe points to 5 when bicubic is used.

Turge08, 2021-12-09:
BTW, this should now be fixed.

closed 2021-12-09 by Turge08