Turge08/print_area_bed_mesh #5: Probe Count Issue using Bicubic with Long Narrow Print Size

frankbags, 2021-11-26:
I have a model, 175 x 30, in a 90 degree orientation and klipper throws the following error 

> !! bed_mesh: invalid probe_count option when using bicubic interpolation. Combination of 3 points on one axis with more than 6 on another is not permitted. Configured Probe Count: 3, 9 

I can re-orient the piece to make the print area more square. 
I don't know the exact limitations or the best way to fix something like this, whether it's capping the longer axis or increasing the shorter axis.

Turge08, 2021-11-27:
According to the klipper source:
<pre>        elif params['algo'] == 'bicubic' and min_probe_cnt < 4:
            if max_probe_cnt > 6:
                raise error(
                    "bed_mesh: invalid probe_count option when using bicubic "
                    "interpolation.  Combination of 3 points on one axis with "
                    "more than 6 on another is not permitted. "
                    "Configured Probe Count: %d, %d" %
                    (self.mesh_config['x_count'], self.mesh_config['y_count']))
            else:
                logging.info(
                    "bed_mesh: bicubic interpolation with a probe_count of "
                    "less than 4 points detected.  Forcing lagrange "
                    "interpolation. Configured Probe Count: %d, %d" %
                    (self.mesh_config['x_count'], self.mesh_config['y_count']))
                params['algo'] = 'lagrange'</pre>

bicubic only allows a minimum of 4 probe points. I'll change the minimum to 5 if bicubic is used.

Turge08, 2021-11-27:
macro has been updated with the changes. Let me know if you're still having issues with this.

frankbags, 2021-11-27:
Thanks, I'll look into it later today. 

If I understand correctly a 2x2 grid is 4 points so it satisfies the 4 point minimum requirement.

frankbags, 2021-11-27:
I see that you have added a statement for homing the axes. 

I think you should add another statement following the G28 call 

`  {% if printer.z_tilt.applied != true %}
        Z_TILT_ADJUST
    {% endif %}`

Turge08, 2021-12-09:
I don't think the rest (z_tilt_adjust, _quad_gantry_level, homing z afterwards, waiting for bed/nozzle to heat up, etc.) belong in this macro.

closed 2021-12-09 by Turge08

frankbags, 2021-12-13:
I wondered why you had a check to see if the toolhead was homed in the XYZ axis in there.  It made sense for me to check if the Z was level as well as homed.