Turge08/print_area_bed_mesh #6: Update manager says "unknown"

srbjessen, 2021-11-27:
Update manager says Unknown.
What did i do wrong? :-D 

[image]
[image]

Turge08, 2021-11-27:
Please update moonraker.conf with the updated "update manager" settings from the Read Me. Also, note that step #1 needs to be performed before adding these settings.

srbjessen, 2021-11-28:
I have updated moonraker.conf and installed it based on the guide, everything works, except the update manager still shows unknown.

srbjessen, 2021-11-29:
With a "Dirty" install, from editing the buffer size, I can see the "dirty" sign with prompts to soft or hard reset. If i do this, it recovers to buffer = 20, and goes back to (?)unknown.

r0bb10, 2021-12-08:
I can confirm this too, it updates regulary but the version is still unknow

[image]

In the moonraker logs the version is ?, could that be the issue?

```2021-12-08 11:30:50,269 [shell_command.py:_check_proc_success()] - Command (git -C /home/user/klipper_printareabedmesh status -u no) successfully finished
2021-12-08 11:30:50,300 [shell_command.py:_check_proc_success()] - Command (git -C /home/user/klipper_printareabedmesh config --get branch.master.remote) successfully finished
2021-12-08 11:30:50,330 [shell_command.py:_check_proc_success()] - Command (git -C /home/user/klipper_printareabedmesh remote get-url origin) successfully finished
2021-12-08 11:30:51,345 [shell_command.py:_check_proc_success()] - Command (git -C /home/user/klipper_printareabedmesh fetch origin --prune --progress) successfully finished
2021-12-08 11:30:51,379 [shell_command.py:_check_proc_success()] - Command (git -C /home/user/klipper_printareabedmesh branch --list) successfully finished
2021-12-08 11:30:51,409 [shell_command.py:_check_proc_success()] - Command (git -C /home/user/klipper_printareabedmesh rev-parse HEAD) successfully finished
2021-12-08 11:30:51,442 [shell_command.py:_check_proc_success()] - Command (git -C /home/user/klipper_printareabedmesh rev-parse origin/master) successfully finished
2021-12-08 11:30:51,481 [shell_command.py:_check_proc_success()] - Command (git -C /home/user/klipper_printareabedmesh describe --always --tags --long --dirty) successfully finished
2021-12-08 11:30:51,529 [shell_command.py:_check_proc_success()] - Command (git -C /home/user/klipper_printareabedmesh describe origin/master --always --tags --long) successfully finished
2021-12-08 11:30:51,530 [git_deploy.py:log_repo_info()] - Git Repo printareabedmesh Detected:
Owner: Turge08
Repository Name: print_area_bed_mesh
Path: /home/user/klipper_printareabedmesh
Remote: origin
Branch: master
Remote URL: https://github.com/Turge08/print_area_bed_mesh.git
Current Commit SHA: 3a8e762abcbfcf9eb22756b6120f287a0cdfca56
Upstream Commit SHA: 3a8e762abcbfcf9eb22756b6120f287a0cdfca56
Current Version: ?
Upstream Version: ?
Is Dirty: False
Is Detached: False
Commits Behind: 0
2021-12-08 11:30:51,530 [base_deploy.py:log_info()] - Application printareabedmesh: Channel: dev, Need Channel Update: False
2021-12-08 11:30:51,530 [base_deploy.py:log_info()] - Application printareabedmesh: Validity check for git repo passed

Turge08, 2021-12-09:
It should be fixed now that I added a revision number to the version number (which is required by Moonraker)

closed 2021-12-09 by Turge08