Turge08/print_area_bed_mesh #7: [FR] Feature Request

srbjessen, 2021-11-27:
An option to set the area where your purge line is, so it's always included in the Mesh, would be super awesome.

Turge08, 2021-11-27:
I thought about this and would contradict the purpose of this macro. I recommend enabling a skirt. The slicer will take it into consideration when calculating the print area and this will be included in the bed mesh.

frankbags, 2021-11-27:
Are you describing a priming line? You usually push more plastic in those so you don't really have to worry about bed adhesion.

Turge08, 2021-11-27:
> Are you describing a priming line? You usually push more plastic in those so you don't really have to worry about bed adhesion.

I assume that's targeted to me? I guess I am.

For those wanting this feature, how would you envision it would work? You would pass the coordinates of the purge line and it would be added to the mesh? 

[image]

Green - Original print area
Red - Purge line
Blue - Adjusted print area

srbjessen, 2021-11-27:
> I thought about this and would contradict the purpose of this macro. I recommend enabling a skirt. The slicer will take it into consideration when calculating the print area and this will be included in the bed mesh.

I didn't even think of the skirt, i usually use brims, so it wasn't in my head, but Superslicer can do both, so I'll start with that.

srbjessen, 2021-11-27:
> Are you describing a priming line? You usually push more plastic in those so you don't really have to worry about bed adhesion.

Yeah, I was thinking about priming, not purging :-)

srbjessen, 2021-11-27:
> I assume that's targeted to me? I guess I am.
>
> For those wanting this feature, how would you envision it would work? You would pass the coordinates of the purge line and it would be added to the [...]

I suppose something like that, although i guess it would cause trouble with Klipper handling the priming and the slicer handling the print area, unless it's only using a variable on the Max-X-Y, and a fixed Print-min at 0,0. (My priming is bottom left)

frankbags, 2021-11-27:
Yes, It was targeted at srbjessen. 

Like I mentioned, I think you should push more plastic in that priming line so a perfectly level bed isn't necessary. 

I also think that Turge08's picture explains how wasteful building a mesh to include the priming line would be.
The only way to make this efficient would be to fill the points of the mesh where no filament would go with zeros. I think it's doable if you build a mesh the size of the blue line and then trigger probes on the green and red areas.   I just think it's unnecessary

srbjessen, 2021-11-27:
> Yes, It was targeted at srbjessen.
>
> Like I mentioned, I think you should push more plastic in that priming line so a perfectly level bed isn't necessary.
>
> I also think that Turge08's picture [...]

Yeah, the problem was that it hits the bed because the corners are slightly up compared to center. But raising it and increasing the flow would combat that issue. I will work it out that way. Will close the post here, as my original idea was not a viable idea :-D Thanks guys, you do amazing work.

closed 2021-11-27 by srbjessen