Turge08/print_area_bed_mesh #8: Move out of range

issic, 2021-11-28:
Generating Bed_Mesh I'm getting negative values! When using your macro.

14:11:57
Move out of range: 32.000 -2.372 8.000 [0.000]
14:11:57
Move out of range: 32.000 -2.372 8.000 [0.000]
14:11:57
Move out of range: 32.000 -2.372 8.000 [0.000]
14:11:57
bed_mesh: relative_reference_index 17 is (318.00, 145.29)
14:11:57
35 | (32.0, 309.3) | (32.0, 332.3)
14:11:57
34 | (89.2, 309.3) | (89.2, 332.3)
14:11:57
33 | (146.4, 309.3) | (146.4, 332.3)
14:11:57
32 | (203.6, 309.3) | (203.6, 332.3)
14:11:57
31 | (260.8, 309.3) | (260.8, 332.3)
14:11:57
30 | (318.0, 309.3) | (318.0, 332.3)
14:11:57
29 | (318.0, 246.9) | (318.0, 269.9)
14:11:57
28 | (260.8, 246.9) | (260.8, 269.9)
14:11:57
27 | (203.6, 246.9) | (203.6, 269.9)
14:11:57
26 | (146.4, 246.9) | (146.4, 269.9)
14:11:57
25 | (89.2, 246.9) | (89.2, 269.9)
14:11:57
24 | (32.0, 246.9) | (32.0, 269.9)
14:11:57
23 | (32.0, 184.6) | (32.0, 207.6)
14:11:56
22 | (89.2, 184.6) | (89.2, 207.6)
14:11:56
21 | (146.4, 184.6) | (146.4, 207.6)
14:11:56
20 | (203.6, 184.6) | (203.6, 207.6)
14:11:56
19 | (260.8, 184.6) | (260.8, 207.6)
14:11:56
18 | (318.0, 184.6) | (318.0, 207.6)
14:11:56
17 | (318.0, 122.3) | (318.0, 145.3)
14:11:56
16 | (260.8, 122.3) | (260.8, 145.3)
14:11:56
15 | (203.6, 122.3) | (203.6, 145.3)
14:11:56
14 | (146.4, 122.3) | (146.4, 145.3)
14:11:56
13 | (89.2, 122.3) | (89.2, 145.3)
14:11:56
12 | (32.0, 122.3) | (32.0, 145.3)
14:11:56
11 | (32.0, 60.0) | (32.0, 83.0)
14:11:56
10 | (89.2, 60.0) | (89.2, 83.0)
14:11:56
9 | (146.4, 60.0) | (146.4, 83.0)
14:11:56
8 | (203.6, 60.0) | (203.6, 83.0)
14:11:56
7 | (260.8, 60.0) | (260.8, 83.0)
14:11:56
6 | (318.0, 60.0) | (318.0, 83.0)
14:11:56
**5 | (318.0, -2.4) |** (318.0, 20.6)
14:11:55
**4 | (260.8, -2.4) |** (260.8, 20.6)
14:11:55
**3 | (203.6, -2.4)** | (203.6, 20.6)
14:11:55
**2 | (146.4, -2.4)** | (146.4, 20.6)
14:11:55
**1 | (89.2, -2.4)** | (89.2, 20.6)
14:11:55
**0 | (32.0, -2.4)** | (32.0, 20.6)
14:11:55
bed_mesh: generated points
Index | Tool Adjusted | Probe
14:11:55
Generating new points...

[gcode_macro PRINT_START]
variable_parameter_PRINT_MIN : 0,0
variable_parameter_PRINT_MAX : 0,0
gcode = 
    {% set BED = params.BED|default(60)|float %}
    {% set EXTRUDER = params.EXTRUDER|default(190)|float %}
    {% set CHAMBER = params.CHAMBER|default(60)|float %}
	G90                             ; Absolute coordinates
	SET_GCODE_OFFSET Z=0.0          ; Reset the G-Code Z offset (adjust Z offset if needed)
	BED_MESH_CLEAR					; Clear the Bed_Mesh
	CG28                            ; initial home
	PARK                            ; move head out of bed heat, but prep for qgl
	M140 S{BED}                     ; Start Heating Bed
	M117 Heating Bed...
	M190 S{BED}                     ; Wait for the Bed to Heat Up
	;M117 Waiting for Chamber Temp of {CHAMBER}C
	;TEMPERATURE_WAIT SENSOR="temperature_fan chamber_fan" MINIMUM={CHAMBER_TEMP}
	;NOZZLE_CLEAN                    ; Clean Nozzle
	G32 				; Home and QGL before Bed Mesh
	BED_MESH_CALIBRATE PRINT_MIN={params.PRINT_MIN} PRINT_MAX={params.PRINT_MAX}
	PARK				; Park and wait for Extruder
	M104 S{EXTRUDER}
	M117 Heating Extruder...
	M109 S{EXTRUDER}                ; set and wait for nozzle temp
	PRIME_LINE                      ; Purge Line
	G92 E0                          ; Reset Extruder
    M83                             ; use relative distances for extruder
	UPDATE_DELAYED_GCODE ID=clear_display DURATION=10

frankbags, 2021-11-28:
Is your probe offset configured properly?

issic, 2021-11-28:
### OFFSET ###
x_offset: 0.0
y_offset: 23.0

So i think, it is.

frankbags, 2021-11-28:
If the probe offset is indeed correct then I suspect the "variable_buffer: 20" line may be causing issues.

can you move the model farther away from the side of the bed ?

yanfoo, 2021-11-29:
I did see this error once, I resliced my model without a skirt and that did the trick.
That being said I should check my probe offset vs mesh min/max, I may have done it wrong, possibly.

Can the macro ommits the buffer in case of model too close?
It is likely not the correct syntax. I think whats happening is the print size is comparing the user mesh limits then if that checks out the generous buffer is  too much. The user can lower the buffer or extra macro check?
Below is an exemple of my reasoning, the syntax is likely wrong. I would have tested it but I'm currently printing something.

                {% if print_min_y|default(0)|float - buffer % >=  mesh_min_y %}
                    {% set mesh_min_y = print_min_y|default(0)|float - buffer %}
                      {% else %}
                        {% set mesh_min_y = print_min_y|default(0)|float %}
                {% endif %}

srbjessen, 2021-11-29:
I have had this issue as well. It IS in fact the buffer, as i have reduced it to 5 and had no issues since. The file however is not editable directly in mainsail, so i now have a "Dirty" install, which I'm prompted to recover.

issic, 2021-11-29:
When sending smaller prints it did worked well. However when the print was almost full 350x350 bed, this happened. I agree with the users above. Check that there is no minus values or values over max bed size?

yanfoo, 2021-12-01:
Very rough, probably a better syntax but this works on my end.
It checks if the buffer can be used, if its out of range it will just do the full mesh per our mesh config.
I tried a model of a cube that I enlarge to the max dimensions of my bed, prior to this code it would output values my probe could not get to.

                {% if print_min_x|default(0)|float - buffer >=  mesh_min_x %}
                    {% set mesh_min_x = print_min_x|default(0)|float - buffer %}
                    {% else %}
                        {% set mesh_min_x = mesh_min_x %}
                {% endif %}

                {% if print_min_y|default(0)|float - buffer >=  mesh_min_y %}
                    {% set mesh_min_y = print_min_y|default(0)|float - buffer %}
                    {% else %}
                    {% set mesh_min_y = mesh_min_y %}
                {% endif %}

                {% if print_max_x|default(0)|float + buffer <=  mesh_max_x %}
                    {% set mesh_max_x = print_max_x|default(0)|float + buffer %}
                    {% else %}
                    {% set mesh_max_x = mesh_max_x %}
                {% endif %}

                {% if print_max_y|default(0)|float + buffer <=  mesh_max_y %}
                    {% set mesh_max_y = print_max_y|default(0)|float + buffer %}
                    {% else %}
                    {% set mesh_max_y = mesh_max_y %}
                {% endif %}

Turge08, 2021-12-09:
It should be fixed in the latest version.

Turge08, 2021-12-14:
@issic Can you please confirm whether you're still having the issue with the latest update?

closed 2022-02-23 by Turge08