Turge08/print_area_bed_mesh #9: Move out of range

RURon, 2021-11-28:
Hello there!

I also have a problem with move out of range on a large print (that covers most of the 300x300 print area) - I get

Move out of range: 322.348 12.608 3.000 [0.000]

My wild guess being that your code maybe tries to do another grid cell than the possible maximum maybe? I'm using 7x7 mesh grid:

[bed_mesh]
speed: 900
horizontal_move_z: 3
mesh_min: 6,11
mesh_max: 286,294
probe_count: 7,7
fade_start: 1.0
fade_end: 10.0
mesh_pps: 2,2
algorithm: bicubic
bicubic_tension: .2

s0ybean, 2021-12-01:
It's the 20mm buffer. Reduce it

Turge08, 2021-12-09:
It should be fixed in the latest version.

Turge08, 2021-12-14:
@RURon Can you please confirm whether you're still having the issue with the latest update?

RURon, 2021-12-16:
@Turge08 Working beautifully now, thank you!

closed 2021-12-16 by RURon