Turge08/print_area_bed_mesh #12: [Future request] Probe before every print

WickedInvi, 2021-12-04:
Can you make an option to always probe despite what mesh is active?

Turge08, 2021-12-04:
Try adding the parameter FORCE_NEW_MESH=True after PRINT_MAX in your slicer gcode

Turge08, 2021-12-09:
@WickedInvi Let me know if this works for you.

WickedInvi, 2021-12-09:
Not working.

15:02:45 
// print_min: 100.911,94.9121
15:02:45 
// print_max: 129.086,123.088
15:02:45 
// last_bed_mesh: 96.5637,90.5644 133.437,127.433
15:02:45 
// No need to recreate Bed Mesh since it's same as current mesh or smaller

Turge08, 2021-12-10:
Sorry, should've been in your print_start macro:

<pre>BED_MESH_CALIBRATE PRINT_MIN={params.PRINT_MIN} PRINT_MAX={params.PRINT_MAX} FORCE_NEW_MESH=True</pre>

WickedInvi, 2021-12-10:
It's working now

closed 2021-12-10 by WickedInvi