Turge08/print_area_bed_mesh #13: Do I need to delete the bed mesh section in Klicky Probe?

kwijibo12, 2021-12-07:
Hi there,

Doesnt seem to pass the variables or its getting cleared. I followed your guide its still bed meshing the whole build plate.

Do you think the below in the klicky probe macro is interfering with your macro?

Thanks,

# Bed Mesh Calibrate
[bed_mesh]

[gcode_macro BED_MESH_CALIBRATE]
rename_existing:             _BED_MESH_CALIBRATE
description: Perform Mesh Bed Leveling with klicky automount
gcode:

    {% set V = printer["gcode_macro _User_Variables"].verbose %}
    {% if V %}
        { action_respond_info("Bed Mesh Calibrate") }
    {% endif %}

    _CheckProbe action=query
    Attach_Probe

    _BED_MESH_CALIBRATE {% for p in params
           %}{'%s=%s ' % (p, params[p])}{%
          endfor %}

    Dock_Probe

Turge08, 2021-12-09:
@kwijibo12  I have the same in my config. Can you try running this from the Console in Fluidd or Mainsail?
<pre>BED_MESH_CALIBRATE PRINT_MIN=100,100 PRINT_MAX=150,150</pre>
This should probe the selected area.

Turge08, 2021-12-14:
@kwijibo12 Can you confirm whether the command above?

kwijibo12, 2022-01-28:
Yes thanks

closed 2022-02-23 by Turge08