Turge08/print_area_bed_mesh #14: Probing a small area front left of the bed. PRINT_MIN/MAX issue.

WickedInvi, 2021-12-07:
Getting a problem with a small patch on the front left of the bed. Using SuperSlicer.

```
15:52:52 
// print_min: -34.0422,-39.6127
15:52:52 
// print_max: 34.0422,34.0422
15:52:52 
// mesh_min: 25,25
15:52:52 
// mesh_max: 54.0422,54.0422
15:52:52 
// probe_count: 5,5
15:52:52 
// relative_reference_index: 12
15:52:52 
// Generating new points...
15:52:52 
// bed_mesh: generated points
// Index | Tool Adjusted | Probe
```
[image]

Turge08, 2021-12-09:
I had a typo in the command in the slicer start gcode. Can you try updating it with the one from the Read Me?

<pre>PRINT_MIN={first_layer_print_min[0]},{first_layer_print_min[1]} PRINT_MAX={first_layer_print_max[0]},{first_layer_print_max[1]}</pre>

WickedInvi, 2021-12-09:
Works fine now.

closed 2021-12-09 by WickedInvi