Turge08/print_area_bed_mesh #15: Euclid Probe

Killajoedotcom, 2021-12-10:
Love that this includes the configs for Klicky, any future support for Euclid or tips on how to use that instead? Is it as easy as replacing "Klicky" commands with new ones?

Turge08, 2021-12-10:
I can look into it. Looks like the 2 following commands are needed before and after:

# Macro to perform a modified bed mesh calibration
[gcode_macro BED_MESH_CALIBRATE]
rename_existing:    BED_MESH_CALIBRATE_ORIGINAL
gcode:
  M401                           ; deploy Euclid Probe if needed
  BED_MESH_CALIBRATE_ORIGINAL    ; check bed level
  M402                           ; dock Euclid Probe

f1nch87, 2022-11-24:
For me it is not working like that.
I get "Probe triggered prior to movement".

Z_TILT Adjust Leveling works with the euclid before it.