Turge08/print_area_bed_mesh #17: Mesh not right? Issue?

WickedInvi, 2021-12-13:
Have a look at this area that has been probed. The move is in the centre of the bed. The size looks fine but the mesh itself looks to be to the back right

[image]

Turge08, 2021-12-13:
Can you paste the output from the console?

WickedInvi, 2021-12-13:
```
17:44:43 
// Force New Mesh: True
17:44:43 
// print_min: 80.9343,74.9304
17:44:43 
// print_max: 149.066,143.071
17:44:43 
// last_bed_mesh: 80.9343,74.9304 149.066,143.071
17:44:43 
// mesh_min: 75.9343,69.9304
17:44:43 
// mesh_max: 154.066,148.071
17:44:43 
// probe_count: 5,5
17:44:43 
// relative_reference_index: 12
17:44:43 
// Generating new points...
17:44:43 
// bed_mesh: generated points
// Index | Tool Adjusted | Probe
17:44:43 
// 0 | (35.9, 46.9) | (75.9, 69.9)
17:44:43 
// 1 | (55.5, 46.9) | (95.5, 69.9)
17:44:43 
// 2 | (75.0, 46.9) | (115.0, 69.9)
17:44:43 
// 3 | (94.5, 46.9) | (134.5, 69.9)
17:44:43 
// 4 | (114.1, 46.9) | (154.1, 69.9)
17:44:43 
// 5 | (114.1, 66.5) | (154.1, 89.5)
17:44:43 
// 6 | (94.5, 66.5) | (134.5, 89.5)
17:44:43 
// 7 | (75.0, 66.5) | (115.0, 89.5)
17:44:43 
// 8 | (55.5, 66.5) | (95.5, 89.5)
17:44:43 
// 9 | (35.9, 66.5) | (75.9, 89.5)
17:44:43 
// 10 | (35.9, 86.0) | (75.9, 109.0)
17:44:43 
// 11 | (55.5, 86.0) | (95.5, 109.0)
17:44:43 
// 12 | (75.0, 86.0) | (115.0, 109.0)
17:44:43 
// 13 | (94.5, 86.0) | (134.5, 109.0)
17:44:43 
// 14 | (114.1, 86.0) | (154.1, 109.0)
17:44:43 
// 15 | (114.1, 105.5) | (154.1, 128.5)
17:44:43 
// 16 | (94.5, 105.5) | (134.5, 128.5)
17:44:43 
// 17 | (75.0, 105.5) | (115.0, 128.5)
17:44:43 
// 18 | (55.5, 105.5) | (95.5, 128.5)
17:44:43 
// 19 | (35.9, 105.5) | (75.9, 128.5)
17:44:43 
// 20 | (35.9, 125.1) | (75.9, 148.1)
17:44:43 
// 21 | (55.5, 125.1) | (95.5, 148.1)
17:44:43 
// 22 | (75.0, 125.1) | (115.0, 148.1)
17:44:43 
// 23 | (94.5, 125.1) | (134.5, 148.1)
17:44:43 
// 24 | (114.1, 125.1) | (154.1, 148.1)
17:44:43 
// bed_mesh: relative_reference_index 12 is (114.99, 108.99)
17:44:48 
// probe at 35.931,46.931 is z=1.987641
17:44:50 
// probe at 55.463,46.931 is z=1.999594
17:44:52 
// probe at 74.994,46.931 is z=2.016328
17:44:54 
// probe at 94.525,46.931 is z=2.066531
17:44:56 
// probe at 114.056,46.931 is z=2.025891
17:44:58 
// probe at 114.056,66.463 is z=2.071313
17:45:00 
// probe at 94.525,66.463 is z=2.131078
17:45:02 
// probe at 74.994,66.463 is z=2.088047
17:45:04 
// probe at 55.463,66.463 is z=2.090438
17:45:06 
// probe at 35.931,66.463 is z=2.045016
17:45:07 
// probe at 35.931,85.988 is z=2.128688
7:45:17 
// probe at 114.056,105.519 is z=2.150203
17:45:19 
// probe at 94.525,105.519 is z=2.164547
17:45:21 
// probe at 74.994,105.519 is z=2.116734
17:45:23 
// probe at 55.463,105.519 is z=2.162156
17:45:25 
// probe at 35.931,105.519 is z=2.157375
17:45:27 
// probe at 35.931,125.050 is z=2.145422
17:45:29 
// probe at 55.463,125.050 is z=2.128688
17:45:31 
// probe at 74.994,125.050 is z=2.114344
17:45:32 
// probe at 94.525,125.050 is z=2.154984
17:45:34 
// probe at 114.056,125.050 is z=2.128688
17:45:34 
// Mesh Bed Leveling Complete
```

frankbags, 2021-12-13:
what's the size of your bed?  probably more than 180x150?

WickedInvi, 2021-12-14:
230x230 usable area is 230x218

Turge08, 2021-12-14:
Closing this since it looks like it's just an issue with how the smaller mesh is being displayed in Fluidd or Mainsail.

closed 2021-12-14 by Turge08