Turge08/print_area_bed_mesh #29: Klicky does not work

Dragi2k, 2022-03-25:
hi,

i have a voron 2.4 with Klicky and Z-calibrate. Your macro runs Straight to the left front corner without picking ub the probe, and gives than a error : Probe triggered prior to movement

Any idea how to fix this ?

Turge08, 2022-03-25:
@Dragi2k Are you using the latest Klicky macros? Do you have "_Probe_Variables" in any of the klicky cfg's?

Can you also post your print_start macro?

Dragi2k, 2022-03-25:
My files are 4-6 month old, on this time i took all files from the calibrate z repo

nope , i have _user_variable 

[gcode_macro PRINT_START]
gcode:
    {% set bed_temp = params.BED_TEMP|default(60)|float %}
    {% set extruder_temp = params.EXTRUDER_TEMP|default(200)|float %}
    {% set z_adjust = params.Z_ADJUST|default(0.0)|float %}
    {% set CHAMBER = params.CHAMBER|default(15)|float %}
    {% set retract = 10 %}
    BED_MESH_CLEAR                    ; clear mesh
    _CG28                             ; Home the printer
    G90                               ; Use absolute coordinates
    PARKCENTER                        ; Move to center
    M117 Heating..
    #_HEATER_ON
    M117 Chamber fan set
    SET_TEMPERATURE_FAN_TARGET TEMPERATURE_FAN=Gehaeuse TARGET={(CHAMBER+10)}
    M106 S255                         ; set print fan to full speed
    M140 S{bed_temp}                  ; Start bed heating
    M190 S{bed_temp}                  ; Wait for bed to reach temperature
    M117 10Min Soaking..
    G4 P250000    
    G4 P250000                    ; Wait for Soak
    #TEMPERATURE_WAIT SENSOR='temperature_fan nevermore' MINIMUM={CHAMBER}
    M117 Waiting for Temp : {CHAMBER}
    M109 S{extruder_temp}             ; Set and wait for nozzle to reach temperature
    M106 S0                            ; turn print fan off
    QUAD_GANTRY_LEVEL PARK=false
    M109 S{extruder_temp}             ; Set and wait for nozzle to reach temperature
    G28 Z                             ; home Z for nozzle clean
    clean_nozzle                      ; clean nozzle
    G28 Z                             ; home Z with clean nozzel
    CALIBRATE_Z
    #SET_TEMPERATURE_FAN_TARGET TEMPERATURE_FAN=Gehaeuse TARGET=35
    #BED_MESH_PROFILE LOAD="100"
    BED_MESH_CALIBRATE
    #BED_MESH_PROFILE LOAD={bed_temp}   ; load mesh if needed
    #LOAD_MESH_TEMP #BED_TEMPERATURE={bed_temp}
    # Adjust the G-Code Z offset with the Z_ADJUST parameter if needed
    #SET_GCODE_OFFSET Z_ADJUST={z_adjust} MOVE=1
    M117 Intro Line..
    G90                               ; Use absolute coordinates
    G1 Y0 X130 Z5 F12000              ; Move the nozzle to the front and near the bed
    G1 Z0.7 F300                      ; Move the nozzle very close to the bed
    G92 E0.0                          ; set extruder position to 0
    G1 E{retract} F3600               ; extrude retract
    G92 E0.0                          ; set extruder option to 0
    G1 X180 E15.0 F500.0              ; intro line
    G92 E0.0                          ; set extruder Poisson to 0
    G1 X184 F6000                     ; move away from intro line
    M117
    #SET_LED_EFFECT EFFECT=light

Dragi2k, 2022-03-25:
on the klipper_Z_calibration is no _probe_variables

is it possible to use _user_variable ?
the z-calibration uses this to look what type of probe you using

Turge08, 2022-03-25:
Try adding this to your config file as a test to see if it helps:

```
[gcode_macro _Probe_Variables]
gcode:

```
Also, you're not passing the print area parameters to the BED_MESH_CALIBRATE

Dragi2k, 2022-03-25:
i will try it, when my print is done.

i use the code in super slicer, or must i add the parms also in the macro ?

Turge08, 2022-03-25:
You're passing the parameters from your slicer to PRINT_START but then need to pass those parameters to BED_MESH_CALIBRATE:

`BED_MESH_CALIBRATE PRINT_MIN={params.PRINT_MIN} PRINT_MAX={params.PRINT_MAX}`

Dragi2k, 2022-03-25:
ok, i will try

Dragi2k, 2022-03-25:
now he got the klicky, but he mashs all and not onlie the printet part

Dragi2k, 2022-03-26:
aa i found the issue, wrong code in the slicer

Has the script a mesh point variable ?

closed 2022-03-28 by Dragi2k