Turge08/print_area_bed_mesh #39: Internal error bed mesh calibrate

shawnp30, 2022-12-21:
[image]

I don't know what happened. Printer worked fine. I stopped a print. Tried reprinting and got this error.

shawnp30, 2022-12-21:
[image]

Turge08, 2022-12-23:
It looks like you have 2 macros that override bed_mesh_calibrate

Turge08, 2022-12-23:
I hae accepted the merge request by @Dan-Ev which should address this issue. Update your euclid macros to the latest version and remove the BED_MESH_CALIBRATE macro from the euclid.cfg.

Here's some more information: https://github.com/Turge08/print_area_bed_mesh/pull/37

shawnp30, 2022-12-30:
i have updated my configs. I now get this error after i start a print and z tilt adjust. console says my probe is already triggered. probe goes to x endstop and stops.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Dan-Ev, 2022-12-30:
If you're doing multiple probes in one go (e.g. homing Z and then QGL, then bed mesh in START_PRINT macro) without stowing the probe each time you need to wrap these with EUCLID_PROBE_BEGIN_BATCH and EUCLID_PROBE_BEGIN_END.

There is an example START_PRINT on the v5 macros page

shawnp30, 2022-12-30:
After g28 my probe stows then picks back out of dock and does z tilt
adjust. Once that is done it stows the deploys again to mesh. Would I still
need to use end batch. I’m using cura. Not using the start print macro.
Using the start GCode in cura. Do I need to use start print macro instead I
orders for it to work?

Dan-Ev, 2022-12-30:
To stop stowing and then immediately redeploying you would use batch mode.

You might be able to add this in your start gcode in Cura, but I don't use it so wouldn't know if it works. I use a START_PRINT macro in klipper and just call that as the start gcode and then make any changes to that macro.

shawnp30, 2022-12-31:
I dont know what i am doing wrong. I have changed configs to no avail. using batch mode in cura yields no result but works. i am trying to use print bed mesh area. where the probe only probes where printed part is. I can get so far as to z tilt, when it get to bed mesh it deosnt follow bed mesh macro, it just meshes the entire bed 
[image]
[image]

shawnp30, 2022-12-31:
[image]

Dan-Ev, 2022-12-31:
tbh I had the same problem with this macro. It didn't seem that the max/min points were passed to the macro and/or were ignored and the entire bed was meshed.

I now use https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging which works well and has the attach probe commands as a variable in the config.