Turge08/print_area_bed_mesh #40: Can't install the macro through PuTTY

dzac21, 2023-01-27:
[image]
I ran the commands from step 1 after SSH'ing into my pi, but it keeps giving me this error when I trying running ~/print_area_bed_mesh/install.sh

sasho4all, 2023-01-30:
That's because the paths have changed. See: [https://github.com/Arksine/moonraker/pull/491](https://github.com/Arksine/moonraker/pull/491). @Turge08 have to update his "install.sh". 
Meanwhile, you can install it manually by uploading the "print_area_bed_mesh.cfg" file to your config folder and following the instructions after point [2](https://github.com/Turge08/print_area_bed_mesh#2-include-macro-in-your-printercfg).

dzac21, 2023-01-31:
Ah alright thank you, that makes much more sense now. Just tried it and it works. Should I change the path on the update manager as well?

3dprintpt, 2023-03-05:
Aready made a puul request to merge the correcctions for the folder structure.