Turge08/print_area_bed_mesh #49: Won't work. But ChipCE code does.

Boztech3D, 2023-08-17:
I can not seem to get this to work. Although the original by ChipCE does work. I am also using Orca Slicer and not sure what to use for it. Not sure what else to say or upload. Thanks

xenspidey, 2023-11-23:
I am also using Orca Slicer and this is how i got it to work. Added the print_area_bed_mesh.cfg file to my configuration directory then added [include print_area_bed_mesh.cfg] to the printer.cfg file. Then added the Orca Slicer start gcode mentioned in the readme. I just tested it on my sonic pad with Orca Slicer and it worked just fine.

suprchunk, 2024-06-23:
I did as xenspidey says, as those are the directions on the page, and get nothing but "sample size exceeded..."

As soon as I remove the call to this, it works as intended. Something within the "code" is wonky, but I haven't had time to trace it yet. Or it may be that I am missing a step that was also missing from the documentation. As is other things we have had to search for solutions to problems introduced with this.